openfab-lab/toolsDocs #2: Error "statusText"

DewiBrunet, 2018-04-04:
Rien de plus frustrant quand tout est prêt mais que ça veux pas ce lancer... ah si il y a plus frustrant, quand personne sais donner une solution^^

Mon problème : le fichier s'importe correctement mais quand je le lance :
/load error
{ "readyState":0,"status":0,"statusText":"

[image]

J'ai essayé:
- copier-coller sur un nouveau fichier
- test sur Illustrator et Inkscape
- passer en px et en cm
- enlever le texte
- enlever les remplissages

Toujours la même erreur....
Ici https://groups.google.com/forum/#!topic/lasersaur/aLKDNF0yygo
on parle de cet erreur et d'éventuel update à faire mais j'y comprends pas grand chose, un conseil @nicolasdb?

DewiBrunet, 2018-10-16:
Il me semble que c'était en effet le nom du fichier (majuscule ou carac spéciaux)
On a une FAQ Debug??

nicolasdb, 2018-10-16:
Top :) 
Alors si c'est le nom à caractère spéciaux, c'est déjà noté sur la feuille sur la machine. 
Faudra implémenter un jour une version numérique. 
Ceci clos celà.

closed 2018-10-16 by nicolasdb