openfab-lab/toolsDocs #6: html&CSS - catégorie?

nicolasdb, 2018-06-12:
html et css, c'est de l'informatique ou de l'électronique? 
@zuperninja a trouvé ceci qui est pas mal. 
http://maitrisezhtmlcss.com/

ça reprend dans un ordre clair les tutos d'un peu partout sur le net.

zuperninja, 2018-06-19:
C'est de l'informatique -> internet -> langage de programmation

nicolasdb, 2018-09-10:
oui mais Electronique > logique de prog > language de prog.

nicolasdb, 2018-09-12:
Donc on range ça où ?^^

nicolasdb, 2019-09-19:
ok, c'est tranché chérie. 
J'ai mis dans informatique, pour le mostly software. Html, les OS, le réseaux. 
Et electronique, le code de prog lié à du hardware électronique, arduino and shit. 
https://github.com/openfab-lab/toolsDocs/blob/master/Informatics/learning2code.md

Par contre, @zuperninja , le lien que tu partageais est mort.

closed 2019-09-19 by nicolasdb