openfab-lab/toolsDocs #11: Optimisation 3dprint

nicolasdb, 2018-09-10:
en lien avec https://github.com/nicolasdb/nicolasdb.github.io/issues/51
et aussi https://github.com/openfab-lab/openfab/issues/232

## Anet A6
- [x] upgrade firmware with last marlin

## Anet E10
- [x] upgrade machine
https://nicolasdb.github.io/post/upgrade-anet-e10/
- [ ] upgrade firmware with last marlin

## software
- [ ] Profil cura for print from SD
https://nicolasdb.github.io/post/curatuto1/
- [ ] settings and update through git push/pull

### Tools: 
calculateur de hauteur idéale pour les layer
https://www.prusaprinters.org/calculator/#optimallayer

**donc pour fabricator**
M4 = pitch 0.7
use layer of 0.7, 1.4, 2.1, 2.8, always multiple of 0.7

Layer height | Error over 10cm | Number of steps | Step length
-- | -- | -- | --
0.20650000000000002 | 0mm | 59 | 0.0035mm
0.21 | 0mm | 60 | 0.0035mm
0.2135 | 0mm | 61 | 0.0035mm

**So for Anet A6 and E10**
leadscrew 8mm, pitch 2mm

Layer height | Error over 10cm | Number of steps | Step length
-- | -- | -- | --
0.19 | 0mm | 19 | 0.01mm
0.2 | 0mm | 20 | 0.01mm
0.21 | 0mm | 21 | 0.01mm

nicolasdb, 2019-01-18:
@satblip 
houston, on a un problème avec les safety de température. 
2 x maintenant que le print s'arrête et que je recommence depuis le début.
C'est le safety du bed qui se déclenche. 
Comment est-ce qu'on bypass cette safety de marde? Quel cable je coupe? le rouge ou le vert?
[image]

nicolasdb, 2019-01-25:
1. je chauffe le lit à 60° pour la première couche puis descend à 30 pour le reste. J'ai l'intuition que ça passe mieux et réduit un peu le Wrapping. 
1. température de la pièce = confort Tshirt. 
1. ça plante particulièrement systématiquement, mais pas exactement au même moment, quand je fais des grand prints. donc plus grand layer 1, donc plus long tps de chauffe du lit à 60° mais c'est peut-être juste une plus grande plage de probabilité statistique. 

probablement 2 paramètres qui entre en compte, séparément, ça marche mais parfois, les 2 en même tps, ça déclenche. 
- si t° ambiante trop froide, la surchauffe pourrait déclencher quelque chose? l’inertie de chauffe du plateau pourrait dépasser les safety et déclencher. 
ça semble pas être le problème pour des petites pièces 3d.
- mouvement prolongé sur Y, = faiblesse du connecteur et faux contact de la sonde? il a l'air bien en contact.... et tient bien.
 
Là j'ai fait 2 prints de 2h, petite pièce, la durée de chauffe lors du layer 1 est plus courte, on passe rapidement à 30° et proportion d'arrêt non remarquable. (aucun cas de mémoire ni constaté récemment)

la longue pièce qui buguait à chaque fois, je l'ai refaite mais 1 pièce au lieu de 4 en même tps. 
c'est passé. le plantage était récurrent en zone layer 1 du 2eme élément

Conclusion, j'ai envie d'accuser une sensibilité un peu trop élevée de la safety lorsqu'on est à 60° avec un plateau alu de cette dimension, 210x270 au lieu de 200x200 pour les plus commun. Il faudrait pouvoir observer le graph de t° et voir où ça bug en lien avec les autres conditions. J'ai bien observé l'écran, la t° est stable mais ça pique les yeux de regarder ça + de 5min. ^^

satblip, 2019-01-25:
Non, ce qu'il se passe, c'est qu'il doit continuellement délivrer du courant pour maintenir les 60°, ce qui ne devrait pas être le cas. On devrait rajouter un mosfet externe (ou en changer). Pour moi, de ce que je lis la protection fait justement très bien son travail ;)

nicolasdb, 2019-01-26:
Ok  j'ai un MOSFEt quelque part. Je testerai ça

nicolasdb, 2022-10-24:
Bon ben on a acheté une Prusa mini. 
C'est la meilleure optimisation du monde.

closed 2022-10-24 by nicolasdb