openfab-lab/toolsDocs #12: Laser - gestion des chutes

nicolasdb, 2018-09-27:
# BAD BAD BAD!
[image]

## Big nono

@Valonlisbeth 
1. Ne pas laisser des chutes non nettoyée des débris.
2. Faire des tests en plein milieu de zone = zones gâchées = Nooooob = le mal

[image]

3. diviser en plaques faciles a ranger et jeter en petit morceaux les chutes trop petites. 
Et mieux organiser son travail pour faire des tests sur des chutes ou dans un coin. 

[image]

[image]

## Please and Thank you

nicolasdb, 2018-09-27:
Et ici, c'est quoi ça? @milenasonneveld  ? ça sent le fabmob ça. 
Je félicite la personne qui ira faire des essais là dessus. Autant dire, qu'on retrouvera encore cette chute dans 2 ans. 

[image]

zuperninja, 2018-09-27:
[](https://media0.giphy.com/media/ETS2RM9CMK7QY/200w.gif?cid=19f5b51a5bacac3971417935325b45f3)

DewiBrunet, 2018-09-27:
[image]

--> https://github.com/openfab-lab/openfab/issues/155

milenasonneveld, 2018-10-01:
> --> openfab-lab/openfab#155

C'est pas beau ?

Déso, j'aurais pas pu laisser ça 😥
--
Milena Sonneveld

T • +32 (0) 497 253 918
Project Vélo M² <http://velom2.be>