openfab-lab/toolsDocs #13: laser module TTL vs PWM

nicolasdb, 2018-09-27:
Ok, Houston, we have a problem. 

I bought another laser module as safety parts for fabmob. 
and because it's never easy and never work on the first attempt, connectors don't fit.

It's same same as the first (I was thinking)
[image]

But with Eleksmaker, it's a 3pin cable for the same module but like this. 
[image]

The control board is  the "SW-LD62 v2"
And I found this part. 
[image]

> Description:
> This PWM TO TTL laser module is design for EleksMaker series machine and its controller board.
> But it can also be used for other DIY machine which need to change TTL modulation.
> 
> Specification:
> Power port size: 5.5*2.1mm
> Working voltage: DC 12V
> Woring power: >2A

Eleksmaker use the mana board controler. 
[image]

The other is laseraxes board. 
[image]
Seems less good than the other. And the power supply connector is a bit loose, it doesn't match the size of the power supply (2.5mm vs 2.1mm probably, tssss)
But have 2 place to connect laser, depending of the laser power. both 2pins. 
`note: I have a second mana board... ^^`

So my question is simply, "what about unsolder 2pin connector and solder 3pin connect instead?" Right?

nicolasdb, 2018-09-27:
last thing is the first laser module doesn't have the board, it's directly connected to the controler board laseraxe. 
https://www.aliexpress.com/store/product/ZONESUN-Engraving-Machine-Laser-axe-405nm-2500mW-DIY-Desktop-Mini-Laser-Engraver-Laser-Cutter-Etcher-17X20cm/4425009_32894209182.html

nicolasdb, 2018-09-27:
Here's the response from AliExpress shop gktools. 
Hi， Nicolas,
We are very sorry for the delivery delay, I will call the express company, then I will reply you.
The 2p2p laser can used for the control board, if you can change the pin connector.
Sincerely!
Jamie.

nicolasdb, 2018-09-27:
Still a problem with eleksmaker and chinese Arduino + laserweb. 
=> BSOD with windows 7,8,10. 
Bug with ch340 USB chipset.... Working fine with linux. 

What is this ArduinoClone? usb chip CH340x
Original is usb chip FTDI 
https://www.exp-tech.de/plattformen/arduino/4450/arduino-nano

Bon, Bsod, j'ai perdu ma doc sur le fabmob10.

https://www.pihomeserver.fr/en/2018/02/17/raspberry-pi-gravage-laser-cnc-eleksmaker-a3-pro/

nicolasdb, 2018-09-27:
rapidement
https://www.youtube.com/watch?v=oXHmOdObTvw

- download hex not repo from grbl gnea
https://github.com/gnea/grbl

- flash on arduino with Xloader, mastaeasy
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino

- laserweb
200x130
power 1000 vs 255

check EEprom setings via "$$"

- check laser mode "$"+
"32","Laser-mode enable","boolean","Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed."

- fine tune acceleration X,Y 

"120","X-axis acceleration","mm/sec^2","X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps."
"121","Y-axis acceleration","mm/sec^2","Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps."
"122","Z-axis acceleration","mm/sec^2","Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps."

- fine tune step/mm 

"100","X-axis travel resolution","step/mm","X-axis travel resolution in steps per millimeter."
"101","Y-axis travel resolution","step/mm","Y-axis travel resolution in steps per millimeter."
"102","Z-axis travel resolution","step/mm","Z-axis travel resolution in steps per millimeter."

actual 
X: 10mm = 31mm

```
$100, $101 and $102 – [X,Y,Z] steps/mm
Grbl needs to know how far each step will take the tool in reality. To calculate steps/mm for an axis of your machine you need to know:
    The mm traveled per revolution of your stepper motor. This is dependent on your belt drive gears or lead screw pitch.
    The full steps per revolution of your steppers (typically 200)
    The microsteps per step of your controller (typically 1, 2, 4, 8, or 16). Tip: Using high microstep values (e.g., 16) can reduce your stepper motor torque, so use the lowest that gives you the desired axis resolution and comfortable running properties.
The steps/mm can then be calculated like this: steps_per_mm = (steps_per_revolution*microsteps)/mm_per_rev

Compute this value for every axis and write these settings to Grbl.
```

nicolasdb, 2018-09-27:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#quick-guide-to-setting-up-your-machine-for-the-first-time

nicolasdb, 2018-10-16:
## note concernant elekslaser little sebastian
16 Octobre

- il manque des "spacers" et la carte controleur est courbée, so bad on so many levels.
Les spacers en nilon sans dans la boite, ils doivent être installé. 
le circuit semble être dessiné pour se placer à la ligne plus bas.

[image]

[image]

- Check écart entre les profils alu, check la mise à l'équère et le serrage des vis de structure. 
Les petites équères métal côté intérieur ne sont pas serré, aucun.
ça va poser problème dans les transports, vibrations, perte de vis, qui vont se retrouver n'importe où, = risque de bug ou endommagement équippement. 

[image]

En soit, l'écart n'est pas si grave et j'ai même agrandit. Pour éviter les tensions dans les matériaux. 
Lorsque j'ai déserré les vis, Les profils s'écartent, tout seul à cause du bras X qui exerce une pression des roues vers l'extérieur . Bref, ne pas forcer. Mais bien serrer et que tout soit bien d'équerre. 

- bras X pas d'équerre, les roues ne touchent pas ensemble chaque butée max. sur Y. 

LA coupe du profilé alu était le coupable. Lorsqu'on serre, le plexi se plaque contre la coupe et se déforme un peu. Ce qui crée tension et désalignement. 
J'ai ajouté une carte de visite à gauche et ça a rectifié le décalage. YEah. 

[image]

- le cable managment. 
BAD bad, zero confiance dans l'usage sur le long terme

[image]
[image]
[image]

ça devrait être bien là. Vive la colle chaude. 

- tension des courroies, à revoir. 
surtout sur X

Done

- le module laser n'est pas d'équerre, pas droit dans son support
Attention à ça pour la précision du positionnement. Le module doit être bien callé dans le fond de son support et bien vertical. 

[image]

- du jeu sur le chariot X
C'est pas résolu, ça m'énerve, j'vais pas en dormir. 
La solution est d'agrandir les trous de vis pour les roues. Et placer des boulons excentriques. (Eccentric nuts) 
https://www.youtube.com/watch?v=nX0J317l0mY

[image]

Ressource tuto assemblage. pour le futur. 
https://www.instructables.com/id/How-To-Laser-Cutter/

Et planche de travail pour en dessous. 

[image]
https://www.thingiverse.com/thing:1678366

## côté logiciel, 
- la mise à l'échelle des axes. 

> Pulley de 20 tooth
> motor 1.8°
> 16microstep
> courroie GT2 

**Donc 80 steps/mm** à confiruger via les commandes `$100=80 et $101=80`

nicolasdb, 2018-10-16:
## Reste 
- mini bug mais potentiellement dangereux. ⚠️ 
Le laser s'allume quelques secondes lorsqu'on connecte laserweb à la machine. 

Solution temporaire, 
1. Soit allumer la machine = bouton noir ON/OFF, APRÈS avoir fait la connexion logicielle de l'arduino. 
2. Soit activer le bouton poussoir sur le module laser.
Lorsque le module clignote rouge, c'est que le module est en low power pour régler le focus. Et ne sera pas activé lors de la connexion. 
Bouton poussoir blanc en bas à droite, à sa gauche, une led est rouge lorsque le module est sous tension (bouton off), clignote lorsque le mode low power est activé. (bouton on)
[image]

nicolasdb, 2018-10-19:
Bug de moteur qui ne sont pas mis sous tension. 
https://github.com/grbl/grbl/issues/589

Depio, 2018-10-23:
> 
> 
> Ok, Houston, we have a problem.
> 
> I order a second minilaser diode for fabmob but I was not please with the first one and I order the same used by bemaker this time. The machine is arrived, it's mounted but the laser module is lost somewhere.
> Hopefully, I bought earlier another laser module as safety parts.
> and because it's never easy and never work on the first attempt, connectors don't fit.
> 
> It's same same as the first (I was thinking)
> [image]
> 
> But with Eleksmaker, it's a 3pin cable for the same module but like this.
> [image]
> 
> The control board is the "SW-LD62 v2"
> And I found this part.
> [image]
> 
> > Description:
> > This PWM TO TTL laser module is design for EleksMaker series machine and its controller board.
> > But it can also be used for other DIY machine which need to change TTL modulation.
> > Specification:
> > Power port size: 5.5*2.1mm
> > Working voltage: DC 12V
> > Woring power: >2A
> 
> Eleksmaker use the mana board controler.
> [image]
> 
> The other is laseraxes board.
> [image]
> Seems less good than the other. And the power supply connector is a bit loose, it doesn't match the size of the power supply (2.5mm vs 2.1mm probably, tssss)
> But have 2 place to connect laser, depending of the laser power. both 2pins.
> `note: I have a second mana board... ^^`
> 
> So my question is simply, "what about unsolder 2pin connector and solder 3pin connect instead?" Right?

I also got the two lasers (3pin and 4pin - Version).

I want to get it up and running without an external board.
However, I have the problem that I can not disable the laser after connecting to the mains.

I have followed these instructions (https://www.instructables.com/id/Laser-Engraving-on-Your-Prusa-MK3/), but this does not work, nor does the connection of a simple power source and the subsequent removal does not turn off the laser. Only disconnecting the power source switches it off again.

nicolasdb, 2018-10-23:
Ok, waw that's quite a hack. @satblip that could be interesting on the prusa :)
_Sorry, I saw I continued in french. I'm using issue also to take note during my process then I'm trying to summarize into a clean .md for my github page._ 

Logically, if the laser is controlled by using the same command as for the fan, using the yellow wire (TTL)
it should work, yes... 
but carefull, like they said: 
> Some printers will have 12V (or even 24V) and you'll need to step that signal down to 5V, otherwise your laser's control module will burn out.

And I actually made this error because I didn't had the right connector and at some point I remember to put red wire on VCC but I changed  cable (needed different length) and cables red/yellow was inverted and I put 12V on the TTL 5V. And I burned a capacitor. 

Anyway, back to your problem. 
I'm actually not sure about the question. Is it to disable the laser when using the printer to print, right? 
If you don't want to disconnect the power supply, maybe a switch, because how they connected the laser, it's not possible to not send the same signal to the fan and the laser. 
Maybe, 5V is enough to trigger the laser at low power...

Depio, 2018-10-24:
Ohh, you missunderstand me. Sorry.

My question:
I bought the 3 pin laser to operate it as described in the manual.

This revealed the following phenomenon:
I connected the laser to the 12V power supply. As soon as I did this, the laser turned on (as described in the description of the laser, this is also normal .. If you do not need the PWM function, you can connect directly to the power port to motor port (12V, GND) on your control board and leave the PWM port unconnected.)

Now I want to use the PWM signal, so I have, as described in the instructions, connected the cable and let the fan rotate. The laser is still fully on. If I deactivate the fan, the laser dims slowly, but then switches back to 100%. It seems to me that the laser processes the PWM signal, but as soon as there is no current at the TTL / PWM pin, the laser automatically switches to 100% (I will probably have to investigate this further)

I also tested to apply 5V to the TTL / PWM pin and then connect the 12V to turn on the laser. This too was unsuccessful. Turning off the 5V left the laser on.

I did the same test with a simple 5V power source. Here, no change in the brightness of the laser was recognizable, since this is just 5V (100% brightness) or 0V (unfortunately, 100% brightness).

After all this was unsuccessful, I ordered a second laser. Now the 4 pin version. Here was a similar phenomenon, which is why I can almost rule out a technical error on the laser almost. But so far I have no approach to finding a solution.

I hope someone can help, because a certain despair spreads at home currently.

At the supplier / dealer, I am unfortunately unaware of, as this itself apparently has no idea regarding the problem.

nicolasdb, 2018-10-24:
Mmmm, right. 

From what I learned from building, using and repairing the lasersaur, (100W CO2) 
The laser tube need power of course but also the TTL part to control the output. 
TTL could bet setup as turned ON when +5V or turned ON when 0V (well it's more >3V or <3V)
https://nicolasdb.github.io/post/lasertech1/

Which could explain why on your module, it turn ON when fan OFF because maybe it need a High TTL signal to actually turn OFF the laser.... 
I'm guessing here. 
But, I never tried your setup. 
Here, I have tried the conversion from 2p+2p connectors to 3p used with the controller board Mana SE. 
and another module, which is without the driver board and it's activated by an analog 12V ON/OFF directly from the controller board (laseraxe) 
I never tried to apply power from an external source. 

> It seems to me that the laser processes the PWM signal, but as soon as there is no current at the TTL / PWM pin, the laser automatically switches to 100% (I will probably have to investigate this further)

Yes, don't forget about the PWM/TLL signal, it should be a constant voltage but with a modulation of frequency. It pulse. 
And without any control on this frequency, it could be always interpreted as "power ON at 100%"
or Turned off in a low pulse, which could be low enough to stop the fan but not the laser. 
It will not show on a multimeter, You will just see the 5V, for frequency, I think you'll need an oscilloscope to look closely at the signal but try to test with different value of FAN speed.

satblip, 2018-10-24:
Really nice hack to keep on the idea board for the future indeed! thx @Depio !

OldShark, 2018-11-27:
Hi nicolasdb. 
Here I am with the same problem as you. I also bought an LD 2.5w that already comes with the 3-pin controller board. I can not make the laser intensity vary. This is using TTL or PWM. I'll see if tomorrow, I'll capture the image of the oscilloscope so you can see what happens.

nicolasdb, 2018-11-27:
I'm not in the lab for a couple of weeks but I saw something engraved on the main body, it was mentionned if the laser was analog or digital. 
Since the diode is the same, Maybe driver boards are set depending on the use. 
 Mmmm. no. 
 Actually, the one engrave as analog power was without driver board and was controled directly from the main controller board. 

The one I just solder a 3P connector instead of 2P2P worked perfectly (except I burn it afterward by inverting PWM and VCC ^^'

diomark, 2019-01-28:
Hi, I'm attempting to run the prusa mk3 laser engraver hack, but I'm a bit confused. I can't get the laser to power on (anything other then the dim test button) with the fan ttl connection.

My question is - for the 3-pin vcc/ground/ttl lasers, does the ttl get connected to the prusa fan's *yellow* line or the red line? 

(i've tested w/ red power line and cant' get it to work:( asking before I take everything apart again..)

Thanks

nicolasdb, 2019-01-28:
Sorry, I don't have experience on the prusa mk3 hack. 
Don't know If @Depio made it work. Or maybe @OldShark ?

OldShark, 2019-01-29:
Hi diomark. 
Is your laser module the same as the image at the beginning of the topic?

OldShark, 2019-01-29:
And i'm not using the driver that came with my laser. I'm controlling for pwm.

diomark, 2019-01-29:
oldshark - i'm using this one - https://www.banggood.com/EleksMaker-FB03-500-500mW-Blue-Laser-Module-2_54-3P-TTLPWM-Modulation-for-DIY-Laser-Engraver-p-1266178.html?rmmds=myorder&cur_warehouse=CN seems to work fine, although to be honest so far I've only tested it on cardboard and paper.

OldShark, 2019-01-29:
The driver is the same, my LD is 2.5w.

diomark, 2019-01-29:
I just ordered that one too just to play with.. -m

OldShark, 2019-01-30:
This year I want to buy a 15w.

OldShark, 2019-01-30:
Where u from? Here is Brasil, São Paulo

djorimar, 2019-03-11:
bonjour,
je suis en galère grave je 
c’est pas ou brancher mon PWM /TLL? 
Quelqu'un peut m'aider. svp
[image]
[image]
[image]

nicolasdb, 2019-03-12:
tu peux donner un peu plus de détails? 
t'as 2 cartes controleurs? la verte c'est quoi? le driver?

djorimar, 2019-03-12:
La verte c'est là où mon Lazer et brancher directement dessus.

nicolasdb, 2019-03-13:
ok, donc carte noire = controleur, carte verte = driver laser. 
en toute logique, la puissance est fournie alors par le driver. donc le controleur ne doit envoyer que de l'info TTL/PWM en 5V. 
Donc, je pense que tu dois connecter ceci: 
[image]

à celà: 
[image]

Il est tjs conseiller de suivre aussi les polarités +- si c'est mentionné sur la carte.

djorimar, 2019-03-13:
Je viens d'effectuer les branchements que vous venez de me conseiller, j'ai
lancé une gravure et le laser ne s'allume pas

OldShark, 2019-03-13:
Bonjour Djorimar et Nicolasdb.
Pardon pour ingérence et pour mon pauvre Français.
Nicolas D'après ce que je comprends, vous utilisez un contrôleur keyestudio, est-ce tout?
Je pense qu'il a déjà une sortie pour contrôler le laser par PWM, mais je ne sais pas quelle est la tension, mais d'après ce que je vois devrait être 12V même. Dans ce cas, vous devrez le connecter à un pilote laser, tel que la carte verte.
Certaines choses Quelle est la tension de fonctionnement de votre laser? Avez-vous un multimètre?

djorimar, 2019-03-13:
Le français est bon merci. oui je pense aussi . Parque quand j'ai acheté la
carte il y avez écrit  .Oui j'ai un multimètre .

"Nouvelles fonctionnalités de la version v0.9 :

Le Baud Rate en série par défaut est désormais de 115 200. (au lieu de
9600)
L'entrée limite de l'axe Z en D11 a été échangée avec une broche permettant
à D12 de prendre en charge une sortie variable de PWM.
Pas de file d'attente : La file d'attente a été retirée car elle est
devenue redondante. La mise en attente met en pause Grbl et n'autorise que
les commandes en temps réel. Le début du cycle reprend, et réinitialise les
sorties.
Nouvel algorithme par étapes fluide : Refonte complète de la manipulation
du pilote d'étapes pour simplifier et réduire le temps de tâche par timing
d'ISR.
Mises à jour de la stabilité et la robustesse.
Planification 4x plus rapide.
Vitesse de rotation de sortie variable.
Compilable via Arduino IDE.
Refonde de l'analyseur syntaxique de la programmation de commande
numérique : Entièrement réécrit pour une conformité totale à la
programmation de commande numérique standard.
Paramètres d'accélération et de vitesse indépendants.
Limites douces : Vérifie si chaque commande de mouvement dépasse les
limites de l'espace de travail avant de l'exécuter, et déclenche une alarme
en cas de franchissement d'un seuil.
Sonde : Les commandes de programmation numérique standard de sondes des
G38.2, G38.3, G38.4 , Et G38.5 sont désormais prises en charge et
connectées via la broche A5.
Décalages de la longueur des outils.
Performances Arc améliorées.
Nouveau simulateur Grbl : Un wrapper complètement indépendant du code
source principal de Grbl qui peut être compilé en un exécutable sur un
ordinateur. Pas besoin d'Arduino. Il simule les réponses de Grbl comme s'il
était sur un Arduino. Cartographie des broches du CPU.
Rapports d'état configurables en temps réel.
Retour à la position initiale mis à jour : Positionne le volume de l'espace
de travail dans un espace négatif, quelle que soit la position limite de
retour.

Contenu :
1 x Carte mère CNC keyestudio
1 câble USB
Détails & caractéristiques

   - Il s'agit d'une carte mère de machine à commande numérique conçue pour
   divers projets DIY et utilisation en usine
   - Réalisez des projets de gravure au laser, de robots d’écriture et de
   dessin de machines Oddball, etc.
   - Dispose de toutes les interfaces nécessaires pour un prix raisonnable,
   peut-être connectée à un disque dur externe
   - microprocesseur : MEGA328P / tension d'entrée : 12 V CC/prend en
   charge les formats de fichier  : Gcode
   - Scannez le code QR sur l'image pour obtenir le tutoriel d’explication,
   le schéma de câblage, et comment installer des logiciels et tester la
   planche

OldShark, 2019-03-13:
Parfait, mais j'aimerais vraiment savoir quel est le problème. Je pensais que j'essayais de connecter la sortie du keyestudio à ce pilote laser. Je ne sais pas si cela fonctionnerait. Cette sortie laser de keyestudio doit être une MLI, car j’ai vu qu’elle pouvait aussi commander un moteur à courant continu.

djorimar, 2019-03-13:
Mon problème c'est que je voudrais faire du nuances de gris.

OldShark, 2019-03-15:
J'ai compris. Et avec cette plaque photo vous ne pouvez pas?

OldShark, 2019-03-15:
En fait, TTL et PWM seront en niveaux de gris. L'un se fera par modulation d'impulsion et l'autre par intensité laser. Je l'utilise par PWM. Le pilote de mon laser étant gravé (TTL), je l’ai fait en utilisant PWM avec la broche 7 d’Arduino. Je fais varier l'intensité du laser, variant ainsi le degré de combustion du matériau.

facumaster, 2019-04-30:
> I got it working, for others that may search here, it is indeed the red fan line not the yellow. The important part was adding a 1k resister between ground and ttl on the laser itself.. (I had initially added a 4.7k resistor which didn't work..)

ive already tried this but nothing, laser still not working. u made it work just adding the resistor??

kdm6182, 2019-08-28:
Help!!!! I'm new to laser engraving, I bought a 5.5w laser with ttl and mounted it on my anycubic i3 mega i connected the fan cable to the ttl and power to independent 12v supply. The ttl did not work at all. Then I connected the fan cable to the 12v input on the laser. It modulated the laser for a brief moment then turned off completely. Can any one suggest what's going wrong? Thanks

OldShark, 2019-08-28:
> Help!!!! I'm new to laser engraving, I bought a 5.5w laser with ttl and mounted it on my anycubic i3 mega i connected the fan cable to the ttl and power to independent 12v supply. The ttl did not [...]

Hello kdm. 
I don't know if this printer is based on arduino mega. What card does it use? You turned on the fan on ttl, you say the laser driver? Have phot of this wire to see? Where u from?

kdm6182, 2019-08-29:
Here is the link to the board on my printer. https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F302581784739
I connected the laser to separate 12v supply and connect the ttl to the fan power and tried to manipulate the pwm of the fan through gcode but the laser stays on at full power. Can't provide photos at the moment. Any ideas?

OldShark, 2019-08-30:
Hi Kdm.
I'm coming home now and couldn't see anything today. Much work.
Very well. Let´s go!
I think the problem is probably because FAN is not PWM controlled. Not sure, because I never used or want to see this 3D. Mine was the one I made and based on Arduino Mega. I believe the FAN control is only On / Off, to cool the extruder heatsink or PLA.
Ru have a multimeter or oscilloscope? 
It would be interesting to actually measure the voltage being sent to your laser. Aliais saw no answer to the question of being directly connected to the laser or its driver. I don't think it's good for you to do this kind of test using the laser, especially using the driver. You can also instead use your laser module, use a piece of LED strip (12V), it will be the same because it is a diode. With the LED strip you can visually check the intensity change without danger.

[image]

The problem I see with using your 3D is not having access to the board firmware.
But back to your 3D. You can achieve grayscale only with On / Off state, but you would have to do so by varying the laser travel speed. It will not be simple to calibrate. Because the longer the laser lasts at one point, the more it will burn the material. If you have one of the equipment I mentioned earlier, test with your G-CODE to see what voltage variation you are getting and then post here.

And where u from? Cause u from Brazil, more simple to help u.

kdm6182, 2019-08-30:
Hi I am in the UK. I will try the led trick. The fan must be pwm as you can directly control the fan speed from the 3d printer when it's printing. I have tested the fan cable on a multimeter and it reads 12v unconnected but as soon as I plug it into either the 12v or ttl port on the laser module  it drops to 2.7v. It's very frustrating. Hope you can help.

OldShark, 2019-08-30:
Oh My Gosh! I´m have the same problem with my laser module. 
[image]

And i thinking i had burned my driver. LoL. Using a mosfet n to control then power by TTL.
[image]

In my case i only remove the laser driver and wire mosfet pwm to laser direct. My TTL pulse 
comes from the pin 7 or 11 in arduino. only this. function perfectly. need only calibrate velocity. Realy make a simple table of velocity x material to brun. 

Teste whit the LED and multimeter or oscilloscope.

OldShark, 2019-08-30:
I still can't figure out why this DDP crash when connecting to the driver. and your case is exactly like mine, it drops to 50% of the voltage. If I'm not mistaken I commented here on Git, but no one could help me. This led me to use this solution. Test there and let's keep in touch. After all you are the first person I know with exactly the same problem as me. I don't even know if your driver is this.
But I think it will do what you want.  Good luck and keep me informed of pleaaaaaaase.

kdm6182, 2019-08-30:
I have done some more research on the board on my printer. And your thought on the fan not being pwm was correct. Apparently you can turn on soft pwm in marlin firmware but I'm hesitant to do this in case I mess things up. It's annoying as I replace my old printer with this one as the other nearly caught fire. And that one did have pwm. I was planning on building a cnc frame with arduino driver in the near future so think I will have to focus on that for it to work properly.

OldShark, 2019-08-30:
Got it. The problem is, I don't see any TTL ports on your card really. But if it uses Marlin as firmware, it must be ATMega based. Surely there must be some lost TTL pin. Ideally an oscilloscope to figure it out.
Make a CNC is good idea. I prefer GrBl to this. My CNC go to change to Mach3 now.

sklymenko848667, 2019-11-19:
hi. can you help me please choose correct board.. let me explain my situation...
i have 3d printer and i bought a laser engraver to install on my 3d printer. but ones i received my laser i realized i ordered wrong one. Now i looking upgrade board for my laser... because, my laser has only 2 pins +/- 12v, but my printer which i need connect it has there only 5v (nozzle fan)... so i need board which will inlude 12v in and out and + 5v control... for example ones on my 3d printer fan is On, board will use dc from 12v inpute

sklymenko848667, 2019-11-19:
hi. can you help me please choose correct board.. let me explain my situation...
i have prusa 3d printer and i bought a laser engraver to install on my 3d printer. but ones i received my laser i realized i ordered wrong one. Now i looking upgrade board for my laser... because, my laser has only 2 pins +/- 12v, but my printer which i need connect it has there only 5v (nozzle fan)... so i need board which will inlude 12v in and out and + 5v control... for example ones on my 3d printer fan is On, board will use dc from 12v inpute

OldShark, 2019-11-19:
Hi sklymenko.
Read this article and watch the videos
https://endurancelasers.com/everything-about-ttl-and-laser-control/