openfab-lab/toolsDocs #15: Auto leveling on anet A6

Jorgininho, 2018-10-24:
Just an isue to trace what I'm doing on the Anet A6 for the autoleveling:

- First we've needed to print a support to fix the detector to the nozzle : 
https://www.thingiverse.com/thing:2205251

- Then we get the current firmware of the anet through the [Git](https://github.com/openfab-lab/Marlin.git)
This is where the fun started....
So 1rst thing, it was more useful to use Platformio (maybe a tuto for that part is indicated to allow future firmware change by more people)
The advantage of platformio is that the environement detect and install by itself package needed, when arduino need manual install for each package.

Firmware need updates, several lignes will be uncommented : 
- to enable auto-leveling : 

> //#define AUTO_BED_LEVELING_3POINT
> //#define AUTO_BED_LEVELING_LINEAR
> #define AUTO_BED_LEVELING_BILINEAR
> //#define AUTO_BED_LEVELING_UBL
> //#define MESH_BED_LEVELING

- to define the position of the sensor regarding the nozzle : 

> #define X_PROBE_OFFSET_FROM_EXTRUDER -10  // X offset: -left  +right  [of the nozzle]
> #define Y_PROBE_OFFSET_FROM_EXTRUDER -60  // Y offset: -front +behind [the nozzle]
> #define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]

- to define which type of sensor we use for autolevel: 

> #define FIX_MOUNTED_PROBE

*****
At this point another issue shows off, by uncommenting these lines, the firmware became to big in size to be uploaded on the command card. A solution is to comment other line about options that we can disable.[
page which describes the space for each options](https://github.com/MarlinFirmware/Marlin/issues/7190)
lines that were commented (need to be confirmed when i'll submit the pull request), also the option choosen might not be the best one: 
- in configuration.h

> //#define PREVENT_COLD_EXTRUSION

- in configuration_avh.h

> //#define AUTOTEMP

From there the firmware can fit.
*****
Here start the destroy process, the upload of the firmware failed, and bricked the card and make it useless because the bootlader got corrupted.
From this point it was needed to flash a new bootloader on the card.
Following this [procedure](https://www.instructables.com/id/HOW-TO-FIX-ANET-BRICKED-BOARD-USING-AN-ARDUINO-UNO/) , and this [video](https://www.youtube.com/watch?v=wauNxu3TjyY)
- we've used a custom cable made by Nico 
- don't forget to charge ArduinoISP script on the arduino used as ISP.
- then we can upload the marlin firmware

to be continued...

Jorgininho, 2018-10-30:
So autoleveling is done on the Anet A6 here are the steps to reach it : 

- The lines that have been commented are the following : 
has been uncommented (related to previous issue comment above) in config_adv.h
> #define AUTOTEMP
commented in config_adv.h
> #define ARC_SUPPORT

We've uncommented the following line because the autolevel captor isn't above when homing the X and Y axis, so this allow a safe Z-homing and avoid the nozzle to press on bed, so in config.h:
> #define Z_SAFE_HOMING

From this point we got a compatible firmware that can be uploaded on the anet a6.
You can Upload it on the Anet

************

## Connect the captor
Next step after mounted the connector on the nozzle with the 3D shape printed above, is to connect the captor to the printer motherboard.
The type of captor used is : SN04-N 
It is a capacitor captor which means it'll detect metalic part as the bed of the printer, putting a glass on the bed may prevent the good use of this captor.
3 wires to connect from this captor to the mother board : 
- The brown wire : alimentation of the captor, needs to be connected on the +12V of the mother board of the Anet (the alimentation connection from the endstop plug doesn't provide enough voltage to allow the captor to works correctly)
- The black wire : the ground, can be connected on the ground of the endstop plug (same as the black wire from original endstop of the anet)
- The blue wire : the signal wire, to be connected on the signal of the endstop plug (same as the red wire from original endstop of the anet)

*****************
## Configuration

Here comes the fun, the configuration of the Zoffset, all command are inspired by [this page](https://3dprint.wiki/reprap/anet/a8/improvement/autobedleveling) , the method used is for the marlin firmware 1.1.x
The parameter that handle the autolevel configuration is the Z offset (M851 in G-code)

All the configuration are done via G-code command via the USB cable directly to the printer, we used [pronterface](http://www.pronterface.com/) to do it (one software is lauched, set baudrate to 250000 before connect)

- 1rst we set the existing Z offset parameter to 0
> M851 Z0
- Then we do homing
> G28
- we move the nozzle to the middle 
> G1 X110Y110
- we move the nozzle so it grip a paper as it should do for a normal functioning, might be possible that the nozzle doesn't want to go too low in poistion, this is due to software endstop that prevent the nozzle to go down if the Z endstop is activated in that case deactivate the software autostop : 
> M211 S0
move the nozzle to the correct position
IF you deactivated the endstop, REACTIVATE THEM !
> M211 S1

- Mark the position to Zero
> G92 Z0
- Use the sensor to probe the middle of the bed (where we just set Z0)
> G30 X110Y110
- Take the Z value of the bed and put a minus sign in front; this is the Z-offset. E.g. 1.23 becomes -1.23
- Set the Z-offset
> M851 Z-1.23
- Store it to the EEPROM
> M500

From now, the autolevel is set up, 2 ways to use it for printing:
- add the autobedleveling code (G29) at the beginning of every jobs done in your slicer (ex: cura) 
or
- do the autoleveling once via command line (the same : G29) and store it to the EEPROM of the printer via command line : M500

The best way to test the configuration is to pring a 1rst layer form like [this one](https://www.thingiverse.com/thing:2741800)

!!!!This configuration may needs to be done each time a manipulation on the printer leads to a movement of the captor from the nozzle (this is why it needs to be fixed RIGID) 

I'll submit the pull request later on, cheers !

nicolasdb, 2018-10-31:
Connect the sensor not captor. huhu
>"he managed to escape from his captors/sensors two nights later"

Jorgininho, 2018-11-01:
So apparently the sensor isn't set-up correctly.
For a reason I ignore, the nozzle was touching the bed during homing.

The behavior of the command G30 needs some investigation since I don't have the feeling it give back results that can be used.
1rst i tried to test the sensor, needs to uncomment the following line in configuration.h : 
> #define Z_MIN_PROBE_REPEATABILITY_TEST

These leads to multiple fail for compiling the firmware as it is (and multiple bricking of the anet :/ )
So I did the test from the firmware with the lines about autoleveling commented.
the result as it is : 

> >>> M48 P50 X100 Y100 V4
> SENDING:M48 P50 X100 Y100 V4
> M48 Z-Probe Repeatability Test
> Positioning the probe...
> Bed X: 100.000 Y: 100.000 Z: -0.032
> 1 of 50: z: -0.037 mean: -0.0375 sigma: 0.000000 min: -0.037 max: -0.037 range: 0.000
> 2 of 50: z: -0.040 mean: -0.0388 sigma: 0.001250 min: -0.040 max: -0.037 range: 0.003
> 3 of 50: z: -0.037 mean: -0.0383 sigma: 0.001179 min: -0.040 max: -0.037 range: 0.003
> 4 of 50: z: -0.037 mean: -0.0381 sigma: 0.001083 min: -0.040 max: -0.037 range: 0.003
> 5 of 50: z: -0.032 mean: -0.0370 sigma: 0.002449 min: -0.040 max: -0.032 range: 0.008
> 6 of 50: z: -0.037 mean: -0.0371 sigma: 0.002244 min: -0.040 max: -0.032 range: 0.008
> 7 of 50: z: -0.037 mean: -0.0371 sigma: 0.002082 min: -0.040 max: -0.032 range: 0.008
> 8 of 50: z: -0.035 mean: -0.0369 sigma: 0.002073 min: -0.040 max: -0.032 range: 0.008
> 9 of 50: z: -0.037 mean: -0.0369 sigma: 0.001964 min: -0.040 max: -0.032 range: 0.008
> 10 of 50: z: -0.040 mean: -0.0372 sigma: 0.002077 min: -0.040 max: -0.032 range: 0.008
> 11 of 50: z: -0.035 mean: -0.0370 sigma: 0.002083 min: -0.040 max: -0.032 range: 0.008
> 12 of 50: z: -0.035 mean: -0.0369 sigma: 0.002073 min: -0.040 max: -0.032 range: 0.008
> 13 of 50: z: -0.035 mean: -0.0367 sigma: 0.002053 min: -0.040 max: -0.032 range: 0.008
> 14 of 50: z: -0.032 mean: -0.0364 sigma: 0.002259 min: -0.040 max: -0.032 range: 0.008
> 15 of 50: z: -0.035 mean: -0.0363 sigma: 0.002211 min: -0.040 max: -0.032 range: 0.008
> 16 of 50: z: -0.035 mean: -0.0363 sigma: 0.002165 min: -0.040 max: -0.032 range: 0.008
> 17 of 50: z: -0.037 mean: -0.0363 sigma: 0.002121 min: -0.040 max: -0.032 range: 0.008
> 18 of 50: z: -0.035 mean: -0.0363 sigma: 0.002083 min: -0.040 max: -0.032 range: 0.008
> 19 of 50: z: -0.035 mean: -0.0362 sigma: 0.002047 min: -0.040 max: -0.032 range: 0.008
> 20 of 50: z: -0.035 mean: -0.0361 sigma: 0.002012 min: -0.040 max: -0.032 range: 0.008
> 21 of 50: z: -0.035 mean: -0.0361 sigma: 0.001978 min: -0.040 max: -0.032 range: 0.008
> 22 of 50: z: -0.035 mean: -0.0360 sigma: 0.001945 min: -0.040 max: -0.032 range: 0.008
> 23 of 50: z: -0.035 mean: -0.0360 sigma: 0.001914 min: -0.040 max: -0.032 range: 0.008
> 24 of 50: z: -0.035 mean: -0.0359 sigma: 0.001884 min: -0.040 max: -0.032 range: 0.008
> 25 of 50: z: -0.035 mean: -0.0359 sigma: 0.001855 min: -0.040 max: -0.032 range: 0.008
> 26 of 50: z: -0.032 mean: -0.0358 sigma: 0.001933 min: -0.040 max: -0.032 range: 0.008
> 27 of 50: z: -0.032 mean: -0.0356 sigma: 0.001995 min: -0.040 max: -0.032 range: 0.008
> 28 of 50: z: -0.035 mean: -0.0356 sigma: 0.001962 min: -0.040 max: -0.032 range: 0.008
> 29 of 50: z: -0.035 mean: -0.0356 sigma: 0.001931 min: -0.040 max: -0.032 range: 0.008
> 30 of 50: z: -0.032 mean: -0.0355 sigma: 0.001979 min: -0.040 max: -0.032 range: 0.008
> 31 of 50: z: -0.035 mean: -0.0355 sigma: 0.001949 min: -0.040 max: -0.032 range: 0.008
> 32 of 50: z: -0.032 mean: -0.0354 sigma: 0.001987 min: -0.040 max: -0.032 range: 0.008
> 33 of 50: z: -0.035 mean: -0.0354 sigma: 0.001958 min: -0.040 max: -0.032 range: 0.008
> 34 of 50: z: -0.032 mean: -0.0353 sigma: 0.001989 min: -0.040 max: -0.032 range: 0.008
> 35 of 50: z: -0.032 mean: -0.0352 sigma: 0.002015 min: -0.040 max: -0.032 range: 0.008
> 36 of 50: z: -0.035 mean: -0.0352 sigma: 0.001987 min: -0.040 max: -0.032 range: 0.008
> 37 of 50: z: -0.035 mean: -0.0352 sigma: 0.001961 min: -0.040 max: -0.032 range: 0.008
> 38 of 50: z: -0.032 mean: -0.0351 sigma: 0.001982 min: -0.040 max: -0.032 range: 0.008
> 39 of 50: z: -0.032 mean: -0.0351 sigma: 0.002001 min: -0.040 max: -0.032 range: 0.008
> 40 of 50: z: -0.032 mean: -0.0350 sigma: 0.002016 min: -0.040 max: -0.032 range: 0.008
> 41 of 50: z: -0.032 mean: -0.0349 sigma: 0.002028 min: -0.040 max: -0.032 range: 0.008
> 42 of 50: z: -0.032 mean: -0.0349 sigma: 0.002038 min: -0.040 max: -0.032 range: 0.008
> 43 of 50: z: -0.032 mean: -0.0348 sigma: 0.002046 min: -0.040 max: -0.032 range: 0.008
> 44 of 50: z: -0.035 mean: -0.0348 sigma: 0.002022 min: -0.040 max: -0.032 range: 0.008
> 45 of 50: z: -0.030 mean: -0.0347 sigma: 0.002123 min: -0.040 max: -0.030 range: 0.010
> 46 of 50: z: -0.032 mean: -0.0347 sigma: 0.002124 min: -0.040 max: -0.030 range: 0.010
> 47 of 50: z: -0.032 mean: -0.0346 sigma: 0.002125 min: -0.040 max: -0.030 range: 0.010
> 48 of 50: z: -0.032 mean: -0.0346 sigma: 0.002125 min: -0.040 max: -0.030 range: 0.010
> 49 of 50: z: -0.032 mean: -0.0345 sigma: 0.002123 min: -0.040 max: -0.030 range: 0.010
> 50 of 50: z: -0.032 mean: -0.0345 sigma: 0.002121 min: -0.040 max: -0.030 range: 0.010
> Finished!
> Mean: -0.034500 Min: -0.040 Max: -0.030 Range: 0.010
> Standard Deviation: 0.002121

seems the sensor is working correctly

to be continued

Jorgininho, 2018-11-02:
So the issue I have (probe offset measured by G30 isn't relevant) already happens in this
https://github.com/MarlinFirmware/Marlin/issues/7662
But when I tried the proposed solution i led to some error in movement command....
Needs to measure the distance between nozzle and sensor manually.

Jorgininho, 2018-11-05:
So I abandonned the use  of the G30 command to measure the offset between the nozzle and the sensor.
The method used to set up the autoleveling is the following : 
> M851 Z0 (set up of the offset to 0)
> G28 (homing)
> M211 S0 (turn off software endstop)
> adapt the height to the nozzle to have to normal setup (just a thin paper that goes through)
> M114 (check the distance in Z after the set up above)
> > if Z=-1.4 then the Z offset needs to be -1.4 mm

> M851 Z-1.4 (will set up the Z offset to -1.4mm)
> M500 (store the Z offset setup)
> G28 (homing)
> G0 Z0 (check that the nozzle is at a printing distance from the bed)

!!!!!! THE DOUBT AND INVESTIGATION !!!!!!

- the code of auto-leveling (G29) NEEDS to be declared in the slicer when generating the Gcode of the job. Needs to be declared after the G28 (homing) otherwise G28 erase G29 measurement.

I leveled 1rst manually the bed, after printing a 1rst layer test, It doesn't seems the autoleveling is taken in account....
I'm king of lost...

Jorgininho, 2018-11-05:
this current issue is like the following
https://github.com/MarlinFirmware/Marlin/issues/5578

There might be an hardware issue where the X axis is tiltedHas a little height difference on Z axis, set it up and launching print of the first layer
same shit...

A mechanical check shows the sensor isn't at the same level of the bed from end to another of the X axis. the autobedleveling is suppose to correct this issue, the weird behavir is that when the sensor is closer to the bed, the correction in Z seems higher than where the sensor is further from the bed

Jorgininho, 2018-11-09:
So the issue seems not software but hardware, the rod of the X axis seems to be twisted regarding the behavior of the compensation from the autoleveling:
https://github.com/AnHardt/Marlin/issues/84

So might needs to correct the X-axis like this : 
https://3dprinterwiki.info/dealing-with-bent-rods/
Or changing the frame of the Anet.

Meanwhile I put back the Firmware as it from actual branch of anet A6 github, but change one parameter to uncomment the following line : 
> #define Z_SAFE_HOMING
So it allows the homing on Z with the probe in the middle of the bed.

After homing we want the prob to be at Z=0 to allow a quicker setup of the bed to the following needs to be changed : 
> #define Z_CLEARANCE_DEPLOY_PROBE   0

The usage of the printer is as before, so bed leveling needs to be done after a normal auto homing.

Soon to be coming a detailled manual on how to burn a bootloader on the anet since its the only succes of all this adventure :)

Its not a give up, if there is a possibility to let the printer as it is today, Iwould love to try to fix this X tilted axis issue in December or January when I'll be back...

When this hardware issue will be solved, here are configuration files for marlin than include the autoleveling and that can be uploaded on the anet: 
[Marlin.zip](https://github.com/openfab-lab/toolsDocs/files/2566115/Marlin.zip)

nicolasdb, 2019-01-18:
Et si on changeait la board? Et qu'on en parle plus?

Jorgininho, 2019-01-28:
nope, à mon avis le probleme est mécanique, il se situe à priori sur la barre de métal de l'axe X qui sont tordues

nicolasdb, 2019-01-28:
il n'y avait pas un soucis de capacité de mémoire de la board? 
C'est peut-être moi qui ai la RAM qui bug. ^^

Jorgininho, 2019-03-13:
Concernant la capacité, c'est pas forcement nécéssaire, j'ai commenté differents parametre pour gagner de la place:
dans configuration_adv.h:

> 	//#define AUTOTEMP
>      //#define ARC_SUPPORT
> 	//#define EXTENDED_CAPABILITIES_REPORT

Sinon, j'ai réorienté la barre pour que la déformation soit à l'horizontale, mais ça ne change rien :(

nicolasdb, 2022-10-24:
Autoleveling? 
voilà, plus de problème. 
[image]

closed 2022-10-24 by nicolasdb