openfab-lab/toolsDocs #18: Shutdown button - script

nicolasdb, 2019-01-07:


Jorgininho, 2019-01-07:
1rst try out is with builtin poweroff button function, like said in recal bx manual:
https://github.com/recalbox/recalbox-os/wiki/Add-a-start-stop-button-to-your-recalbox-(EN)
But this only need one click to poweroff, needs to find a solution where button needs to be hold on (5S) to poweroff.

Jorgininho, 2019-01-07:
May have a solution by modifying the scripts /recalbox/scripts/rpi-pin356-power.py
see :
https://github.com/recalbox/recalbox-buildroot/blob/master/board/recalbox/fsoverlay/recalbox/scripts/rpi-pin356-power.py

1rst try, modifying value speed
```
	if channel == POWERPLUS:
		speed=0.15
		shutdownstring="shutdown -h now"
	elif channel == RESETPLUS:
		speed=0.05
shutdownstring="shutdown -r now"
```
to
```
	if channel == POWERPLUS:
		speed=2.15
		shutdownstring="shutdown -h now"
	elif channel == RESETPLUS:
		speed=1.05
shutdownstring="shutdown -r now"

```

hoping to add a delay to 2seconde, but no results

> - ! No sudo mode for recalbox, needs to do a mount/unmount stuff  (Which maybe could be a solution for this issue): 
> https://github.com/recalbox/recalbox-os/wiki/Add-your-own-startup-script-(EN)
> - tip : to acces terminal via emulationstation interface press F4, then ALT+F2. Root credentials are user : root , pass : recalboxroot

Jorgininho, 2019-01-07:
modification on this file weren't taken in account before because the wrong line was uncommented in recalbox/share/system/recalbox.conf
the line to be uncommented : 
> ;system.power.switch=PIN356PUSHRESET 

## The parameter to modify, in /recalbox/scripts/rpi-pin356-power.py : 
in function : 
```
def push_mode(channel):
...
while flag:
...
   if (timer>30):
       ....
   if (timer>10):
       ....
```

So to turn off the console, keep pressing 3sec, to go back to home screen of recalbox, press 1sec

Jorgininho, 2019-01-07:
trying now to have a LED blinking while pressing the button following 
https://github.com/recalbox/recalbox-os/wiki/Add-a-start-stop-button-to-your-recalbox-%28EN%29

on 1rst try, the behavior of the LED is the following : 
- the LED stays ON when console is ON
- It blinks when the system shutdown

nicolasdb, 2019-01-11:
ok, COOL
Donc, le mode d'emploi sera de? combien de sec pour quoi faire? 

Et on clot l'issue

closed 2019-01-15 by nicolasdb