openfab-lab/toolsDocs #21: cover speaker

nicolasdb, 2019-01-07:


nicolasdb, 2019-01-08:
Same biz, scan HP et inkscape. 
[image]

nicolasdb, 2019-01-09:
Ce serai en fait plus simple de tout changer que de chercher à sécuriser l'electrique des HP. 

Donc, j'ai ça en stock. 
TDA7492P Bluetooth 2x25 W Sans Fil V2.1 Bluetooth Audio Récepteur Conseil Module Amplificateur
[image]

> Alimentation: 2.1 interface d'alimentation DC 10 ~ 25V (avec protection de connexion inverse)
> Entrée audio: réception Bluetooth
> Impédance de sortie: 8 ohms maximum (peut piloter des enceintes d'impédance de 4, 6, 8 et 16 ohms)
> Puce d'amplificateur: TDA7492P (avec la lettre P est le double canal 25 watts) 25W + 25W
> Puissance de sortie: 2 * 25W 

+ transfo 12V
+ cable jack-jack 3.5 
ET c'est bouclé. 

Question: Réglage son se fait où?

nicolasdb, 2019-01-09:
[image]

Jorgininho, 2019-01-10:
On va finallement garder l'elec de base des HP, l'ampli proposé est surdimensionné pour ces petits machins.
Je m'en vais finir de démonter les HP et l'elec associée.

nicolasdb, 2019-01-10:
voilà le design qu'on va utiliser. 
Gros boulot de motif répétitif et adaptation au cas présent. 
[image]

nicolasdb, 2019-01-11:
On notera la petite surprise de style lorsqu'on applique le vernis à l'eau sur le hachurage laser dans le MDF. 
La texture créée est trop cool. ça fait un effet un peu écorce d'arbre et redonne du volume à la gravure.  Le touché est très intéressant aussi. 

[image]

_note: j'applique 3 couches en insistant bien sur la gravure._

closed 2019-01-11 by nicolasdb