openfab-lab/toolsDocs #27: Changer encre 2D print

DewiBrunet, 2019-05-10:
## Docu quick & dirty pour recharger l'encre de l'imprimante 2D

! Ça risque de tacher, à ne pas faire en robe blanche

### 1) sortir l'encre qui manque et la seringue dans la caisse en dessous
[image]
[image]

### 2) Percer l'opercule
[image]

### 3) ouvrir le capeau
[image]

### 4) enlever le caoutchou et versser le contenu de l'encre
[image]

### 5) venir aspirer l'excédent d'air avec la seringue dans l'imprimante (regarder pour plus avoir de bulle d'air dans le tube)
[image]

### 6) tout refermer et aller dans Setutp -> Maintenance -> Print Head Cleaning
[image]

### 7) printer un test et recommencer le print head cleaning jusqu'à avoir un truc clean
[image]
[image]

### 8) tout bien nettoyer, mettre le flacon vide au déchet toxique

[image]

nicolasdb, 2019-09-11:
astuce driver aussi. 

[image]

C'est ici:
https://github.com/openfab-lab/toolsDocs/edit/master/Informatics

nicolasdb, 2019-09-11:
Il est où le fichier qui résume tout ça, bien mis en page? 
Il faut en parler sur le RTFM

nicolasdb, 2019-09-19:
Les informations relatives à l'impression 2D papier sont ici: 
https://github.com/openfab-lab/toolsDocs/tree/master/Informatics