openfab-lab/toolsDocs #28: Refurbishing of CNC : GOCNC hobby A4 basic

Jorgininho, 2019-06-17:
Recently, a member of Openfab gave the lab a CNC he wasn't using anymore.
This is a tiny CNC which come in perfectly for the community to get hands on this kind of machines.

With this issue we'll try to write down the refurbishing of this machines, to make sure all person of the comminity that is interested can participate, to organize and make sure all info are accessible for future works on it.

So here are the objectives that have to be done for a simple usage :

- [ ] Install Linux CNC on the machine dedicated to the CNC : 
This machine needs a computer dedicated to it, that will command it during whole time of the jobs.
The computer is unmounted, the trick here is know how the linuxCNC OS needs to be installed ? via USB or CD-ROM ?
[image]
From this step, the computer itself shall be able to command the CNC, first test needs to be done to check everything is alright (command on the 3 axis X,Y, Z)

- [ ] Callibrate steps of the CNC machine
Small task, but mandatory is to callibrate the command of the engines via linuxCNC, this means make sure, for example, that a 1cm movement is a 1cm movement on the CNC

- [ ] Flatening of the bed
Some of the 6 screws of the bed hae their head little above the 0 level, just screw them so the level 0 of the bed is flat everywhere. Not tricky, just a make it works better task.

- [ ] Draw something, Understand setup of a jobs
The actual support on the CNC shall allow mounting of a pen, a good first job/test would be to draw something as it is.
This is also a good opportunity to understand how this machine is setup for a job, for example I don't see any homing end-stop, which is present on almost all 3-axis machines.

- [ ] connect the driller to the CNC
Actually, only the motors for the 3 axis are connected to the CNC, so to make it works completely it is needed to connect the driller. Some research needs to be done to know on which part of the command card we can connect the driller.
[image]

- [ ] Choose the supported driller 
There is actually no dedicated driller for the CNC, 2 solutions are proposed, a Dremel-like Proxxon, or a DC motor.
[image]
For the solution choosen, a dedicated support needs to be designed, either replace the blue support plate already in place or design an adaptator to mount on it (2nd solution prefered because it shall support better the mechanical effort)
[image]

At this state, the CNC shall be usable for classical CNC jobs, what is needed shall be improvements
---------------------
WANTED IMPROVEMENTS

- [ ] Clamping system for the bed
Find a solution to fix material on the bed

- [ ] Replace computer by raspberry Pi
We are using an old computer because of the parallel connector, but it seems possible to use Raspberry Pi GPIO as parallel connector.

- [ ] Y-belt tensor replacement
The beering used to tense the Y-belt seems of axis, should use a different screw with bigger diameter to make it more centered.
[image]

- [ ] Replacement of the command card
The actual configuration of this CNC is the same as the big CNC of Openfab, which make it a great introduction tool to CNC for Openfab members. But this command become an old technology and it should be a good improvement to replace the command card by a most recent one which allows USB connection (and end-stops!). This replacement need to occure then on both CNC to keep both machines accessible to Openfab members.

nicolasdb, 2019-06-17:
[image]

https://www.youtube.com/watch?v=uDB2T3JMJFA

O-Navid, 2019-06-17:
Here we are, Abdel and I are pretending to know what we are doing.
So here it is : we are currently downloading LinuxCNC and will soon install it on a USBfllashdrive callde Martens. Martens is a mainly white Flashdrive and will be our companion for the rest of the day.

We also had to download and install Win32-diskimager to write on that flashdrive (for apparently, just copy-pasting it won't work in this case).

3 minutes to go 'till the completion of the download.

O-Navid, 2019-06-17:
Here we go, this is a very portable laptop. 
We plugged in a screen, an old keyboard and a old mouse.
Such a delight to see those oldies serve again *-*
#nostalgia #vintage #nicoyourenext

[image]

nicolasdb, 2019-06-17:
ha et puis, ça peut être cool comme projet de faire une boite pour le PC controleur. 
https://buildyourcnc.com/AssembledElectronics.aspx

[image]

abdel1130, 2019-06-17:
you have to enter boot with the usb linux graphic download

login: minicnc
passwords: minicnc

abdel1130, 2019-06-17:
[image]

O-Navid, 2019-06-17:
TO DO : 

- install LIMIT switches for X, Y and Z axis
- Emergency STOP knob (find a practical spot on the machine to install it)
- define working area (on X, Y and Z)
- some cable management might be usefull too

After that we have a functionning machine, but with no tool.
Need to fix the spindle (we can use the Proxxon/Dremel : need to find a way to fix it on the machine)

Building a box for the Computer might be nice too.

Where are we going to place the machine ? We can fix it on a IKEA-rollroll-woody-wheely-drawer-table for instance.

nicolasdb, 2019-06-19:
La config temporaire dans linuxCNC stepwizard, _il faudra ajuster pour identifier Estop et XYZ limits_
[image]

Connexion de cette manière. 
[image]

nicolasdb, 2019-06-28:
Bon, puisque ça avance pas et que j'en ai quand même un peu besoin. 
En 2h30, 
j'ai ajouté une boite pour l'ordi, et j'ai ajouté les endstop X et Z. 
Merci à Francis pour avoir installé le Y. 
Reste à cabler et calibrer l'espace de travail. 

### boitier super hightech
[image]
[image]

### switch limit X
[image]
### switch limit Z
[image]
### switch limit Y
[image]

O-Navid, 2019-07-10:
Limit Switch X, Y and Z are now wired to the micro-controller !
We tried connecting the 3 of them to the board, but it was to thick, so we managed another way out of it : switch X and Z are wired to switch Y, who in turn is connected to the board.
Voilà.

nicolasdb, 2019-07-11:
[image]

nicolasdb, 2019-08-09:
Pour les adicts d'adobe. 
y a peut-être ici une solution pour le controle CNC 2D. 
https://diegomonzon.com/illustrator-to-g-code-panel-v2

nicolasdb, 2019-08-19:
## installation fraiche de linuxcnc sur mint19
Pourquoi? parce que le debian d'origine, c'est un peu de la dèche de noyaux. et Inkscape est bloqué à sa version 0.48
https://forum.linuxcnc.org/9-installing-linuxcnc/35627-linuxcnc-on-linux-mint-19

nicolasdb, 2020-01-05:
Note: Leadscrew pitch 3mm. 

## J'ai trouvé un bug
L'axe Z a un backlash de ouf. La vis est ok mais c'est le pasdevis (nut) qui doit avoir prit dans la gueule. Les autres axes sont bon.

nicolasdb, 2020-01-05:
Curieux, c'est pas tant un bug, ya du jeu lorsqu'on monte pour faire le homing Z. J'avais réglé à 60mm le Zmax, j'ai rajouté des planchettes (2x3mm mdf) pour éloigner le contact et réglé à 55mm Zmax. 
ça marche, plus de jeu. 

Maintenant, surfacing. Pour mettre à zero la plaque de bois, (martyr), y a un script pour générer le gcode. 
https://github.com/LinuxCNC/simple-gcode-generators/tree/master/face

nicolasdb, 2020-01-05:
FAQ: 
homing déconne? il faut bien que la machine ne soit pas en contact avec un limitswitch avant de lancer le homing. 
J'ai eu le coup, et Z cherchait vers le bas.

nicolasdb, 2020-01-05:
https://ncviewer.com/

nicolasdb, 2020-01-05:
Shiiiiiiiiiiiit. 
Surfaçage foiré. 
La mèche glisse et descend très progressivement. 

La broche proxxon ne doit pas être assez forte, j'ai pourtant serrer à la pince. 
Je reessayerai avec une mèche droite, ici elle était hélicoidale, mais quand même, j'ai jamais eu ça sur le Kress sur l'autre machine. 

pistes:
https://gocnc.de/shop/de/451/hf-milling-spindle-micro-hornet?c=39
http://www.aliexpress.com/item/4000293295742.html

nicolasdb, 2022-10-24:
[image]
 on a acheté une autre. 

Elle est plus rigide, même si pas encore foufou pour faire des choses de précision dans le 10ème de mm. 

Le système de controle choisi est la black BOX d'openbuilds. 
[image]
https://openbuilds.com/builds/blackbox-motion-control-system.8320/

On passe en GRBL1.1. 

Le même controleur a été installé aussi sur le lasersaur pour uniformiser les controles machines. 
Opensource, merci.

closed 2022-10-24 by nicolasdb