openfab-lab/toolsDocs #34: Laser Module is not on/off acoording to PWM signal

Sashika4950, 2022-03-07:
1)My laser module (D-B 5500) is always on when I connect the PWM signal cable while running the uploaded file.  so it engraves the  movements of the machine.  Should laser module be switched on and switched off according the PWM signal  that we need to engrave or is it signal strength become low and high . I used laserGRBL software.

2) please tell me how to change the machine movement speed from the LaserGRBL software.

nicolasdb, 2022-04-23:
Sorry, I didn't check my github notification. 
And it's been a while since I did some work with the laser module. 
My guess if the module is always on, is you should double check if your module is PWM or analog. 
Since PWM send 5V pulses with a modulation of frequency to send digital power settings, and analog 0-12V will adjust tension to set power. 
If analog receive PWM data, it will receive only continuous 5V and will set the power at a constant ~40%
The other way, PWM receiving 12V, the module will burn out. I learned that the hard way. ^^

Sashika4950, 2022-04-23:
Thank you for your reply. I will study further and get back to you.

closed 2022-09-16 by nicolasdb